VzBoT3D/VzBoT-Vz330 #2: Power supply 48V

demonlibra, 2021-11-08:
Motor Wantai 42BYGHW811
Rate voltage 3.1V
Rate current 2.5A

Maximum power dissipation for this motor 3.1V x 2.5A = 7.75W
Two motors 7.75W x 2 = 15.5W

Why you preffer power supply 350W?

HoshuaK, 2021-11-19:
I believe he's running the stock Tronxy bed which runs on 24V.

demonlibra, 2021-11-19:
I write about power supply 48V which using only for stepper motors.

mouamart, 2021-12-26:
> I write about power supply 48V which using only for stepper motors.

I believe it's to power the TMC5160HV driver's not the stepper motor directly

demonlibra, 2021-12-26:
Just think about it.
I am sure you can use a power supply with a much lower power, for example, 50 watts.
Of course this requires real testing.

mouamart, 2021-12-26:
> Just think about it. I am sure you can use a power supply with a much lower power, for example, 50 watts. This will be a threefold power reserve. Of course this requires real testing.
You're stepper motor problem don't need the 48v psu but the 48v system is for the ODrive motor 

https://youtu.be/1XzD7lGNaek

demonlibra, 2021-12-26:
I thought you were using 48V for stepper motors.
https://www.youtube.com/watch?v=B0E0zaYVU94

tehmaze, 2021-12-27:
For motor power budget use the formula on the Watterott wiki: https://learn.watterott.com/silentstepstick/faq/#what-power-supply-do-i-need

closed 2022-01-06 by tehmaze