VzBoT3D/VzBoT-Vz330 #3: Z stop that hands off bed

coilboii, 2021-11-17:
Is there a stl for the hindge that hangs off the side of the bed with the screw sticking up to hit the z stop or is there a item you used. Nothing in the bom nor the stl

coilboii, 2021-11-17:
Supposed to say hangs off

simonvez, 2022-01-06:
here mate : https://github.com/VzBoT3D/VzBoT/blob/master/Assemblies%20BOM%20and%20STL/Bed%20Assembly/STLs/Z%20height%20adjustement%20screw%20mount.stl

simonvez, 2022-01-06:
if you want to test the new one coming in CU4... I can share you the files now ?

simonvez, 2022-01-06:
[Z switch mount.zip](https://github.com/VzBoT3D/VzBoT/files/7824916/Z.switch.mount.zip)

here you can test it

simonvez, 2022-01-06:
[image]

closed 2022-01-08 by simonvez