VzBoT3D/VzBoT-Vz330 #4: BOM

goosst, 2021-12-04:
Interesting project!
This might not be the best place to ask but:
- I understand there is a BOM on this github site. To make a cost estimate however, it's not clear which parts of this BOM are reused from the tronxy and which need to be upgraded / get thrown away from the tronxy. Is it fair to assume that all extrusions, lead screws, stepper motors, heated bed and support are reused in this build?  
   - Or even better, could it be added to a column in the BOM what is new and what is original to the tronxy?

- any advice where to start? Are there some guides out there or is it just going based on the CAD files?

thanks!

mouamart, 2021-12-26:
[image]
It's in the spreadsheet on top. If it say necessary buy it means this part is not included with tronxy x5sa kit ,  optional buy  Means either the part is included but not very good quality or the features is optional for both new build or tronxy x5sa upgrade. None that means this part is included with tronxy x5sa kit.
Hope that helps

goosst, 2021-12-28:
thanks! 
or I was blind or it has been added since I opened the ticket :)

closed 2021-12-28 by goosst