VzBoT3D/VzBoT-Vz330 #6: Firmware for the build

mouamart, 2021-12-26:
A lots us not to familiar with firmware compiles. Is there going to be a firmware ready to use or a step by step on how to install the firmware for vzbot??..

simonvez, 2022-01-08:
It is a bit hard to provide this. I can provide my klipper config, but its based on my printer. Not everyone runs the same board, drivers, electronics.. so it is very hard to provide all the information. I can however provide what I run if you wish

mouamart, 2022-01-08:
> It is a bit hard to provide this. I can provide my klipper config, but its based on my printer. Not everyone runs the same board, drivers, electronics.. so it is very hard to provide all the information. I can however provide what I run if you wish

Just provide firmware based on BOM and if you know some good YouTube videos and documentation as a reference that would be a bonus

karlspena, 2022-01-08:
Please do

thantik, 2022-01-16:
A klipper config would certainly be a huge jump in progress when I get to that point.  I'm still waiting on my parts from Mellow (CF tube and alum brackets), but I've got the Mellow Super 8 board, and basically the exact same configuration.  Lots of boards actually route a lot of the same wires, so a klipper config will probably even match up with most other boards with the same processor - or at least is a good jumping-off point to modify to fit an existing board.  Macros and such that you're using also give people sane parameters to aim for in order to replicate your work.