VzBoT3D/VzBoT-Vz330 #7: Electronic

mouamart, 2021-12-26:
your using mellow super 8 motherboard as an official board for the build, raspberry pi 3, pitft50, 8 bar relays ext. Can you add wiring diagrams and instructions on how to wire everything together. Thanks in advance

simonvez, 2022-01-08:
the 8bar relay is no longer used in my case since the Super8 has enough fan outputs for our needs.  The wiring is pretty straight forward for the rest.  The Pitft50 is wired with a 1m ribbon cable for pi

mouamart, 2022-01-08:
> the 8bar relay is no longer used in my case since the Super8 has enough fan outputs for our needs. The wiring is pretty straight forward for the rest. The Pitft50 is wired with a 1m ribbon cable for pi

Then probably you should remove it from the bom as I did purchase it based on the bom list. 
I do understand it's straight forward and probably won't take me much to figure it out but a wiring diagram based on the BOM would b helpful for a lots of people as lots of people don't know how to wire a thing without instruction.

simonvez, 2022-01-26:
yes I need to work on this too

simonvez, 2022-01-26:
I will make the relay optional for those running smaller boards

closed 2022-04-11 by mouamart

simonvez, 2022-04-11:
Thanks a ton Mouamart!!!