VzBoT3D/VzBoT-Vz330 #8: Google spreadsheet.

shawnp30, 2022-01-07:
I’m trying to use the Google spreadsheet for the bom to make changes to the build size. Is there a way to open the Googlespread sheet in sheets on iPhone?

simonvez, 2022-01-07:
I think you can get the suite installed no?

chrishenze, 2022-01-07:
Here is the normal link to the spreadsheet, this shouldwork better?
https://docs.google.com/spreadsheets/d/1eNbgQ2HAe7iDxFRou1gHZOtjP1PLiPEpZYigivr-bK0/edit#gid=242794051

tehmaze, 2022-01-07:
Works fine on iPhone with https://apps.apple.com/nl/app/google-sheets/id842849113

[image]

closed 2022-01-07 by tehmaze

fabform, 2022-01-09:
Handy to locate google sheet id
[find google sheet id](https://fabform.io/a/how-to-find-google-sheet-id)