VzBoT3D/VzBoT-Vz330 #9: wanting to adapt to x5sa 400+ models

naruhaxor, 2022-01-11:
what all would be need for an adaptation to the 400 models of the X5sa? i am currently trying to ompile a buy list (i am basically having to (de) pro mine to get rid of the tronxy linear rails but i like where this is going and would also like to possible offer my assistance in creating adaptations to the larger tronxy core xy printers

chrishenze, 2022-01-11:
This answer is on the guess that the frame of the x5sa with 400mm travel is just 70 mm wider than the normal (pls recheck) 

[image]

So you would need a longer cf tube with 510mm, but a normal 500mm should also work and would be easier to source. 
2x MGN12H in 450mm should also barely fit the travel, as well as 450mm MGN9H

naruhaxor, 2022-01-11:
So if I end up just getting the 500 mm extrusion do you have the step files
for the XY carriages that allows it to attach to the sides of the printer
so that way I can adjust if necessary

chrishenze, 2022-01-13:
There is only one printed version of the gantry part. I checked the CAD model and 5 mm shorther on each side might not work, you would have to cut a semicircle ouf ot the cf tube:
[image]

closed 2025-01-02 by pbsuper