VzBoT3D/VzBoT-Vz330 #11: I would like to ask, where to buy or print this accessory, I see that it is in your video but not in the accessories list. thanks

psusong, 2022-01-16:
[image]

chrishenze, 2022-01-16:
These are the oldham couplers and currently you need to print, sand and lube them to get them working. Mellow should make them in the long run.

psusong, 2022-01-17:
I was printing him and saw in the video there was new copper.
[image]

karlspena, 2022-01-17:
Where can we get the brass oldham couplers?

chrishenze, 2022-01-17:
The oldham coupler on the picture is a prototype from Mellow, they will probably sell them in the future.

chrishenze, 2022-01-17:
I added the STLs of the oldham coupler for easier printing.

chrishenze, 2022-01-20:
They are now available from Mellow: https://s.click.aliexpress.com/e/_ASAMpS

closed 2022-01-24 by simonvez