VzBoT3D/VzBoT-Vz330 #14: about pei？

psusong, 2022-01-17:
Hello, can you see how you dealt with the pei, whether it is the magnet placed behind the aluminum plate or the magnetic surface directly attached

chrishenze, 2022-01-17:
The magnetic sheet should be directly put on the heated bed.

psusong, 2022-01-19:
[image]
I see that some people do this. I don't know if it will be deformed. If the magnetic film is directly attached, will it be uneven?

chrishenze, 2022-01-22:
You don't need a magnetic sheet if you already have magnets in your bed. You would just put the spring steel plate with the PEI layer on it.

closed 2022-01-25 by chrishenze