VzBoT3D/VzBoT-Vz330 #15: Provide CAD Nema23 Motor mount

FLXMM, 2022-01-23:
Hi could you plz provide the .step file for the nema23 Motor mount so i can make a belt tension option similar to the nema17 mount for the vz printhead.

simonvez, 2022-01-24:
sure.. but I only have the one with the motor on the side. I think motor has to remain on the side, while there is the same mechanism on the back to pull on the pulleys. And use the same thensioning method with the same hole in the back, and same top cover hinges or brackets

https://github.com/VzBoT3D/VzBoT-Vz330/tree/master/CAD/Gantry/Motor%20Mounts/ODrive-Nema23

closed 2022-01-24 by simonvez

FLXMM, 2022-01-24:
Hi thx... i thought about making a mechanism that moves the whole 10mm shaft including both bearings.
But that might over complicate the design... alternativly i thought about editing the existing tensioning mount and just put the Nema23 upside down on top and tension it that way. that would also get rid of the closed belt, bearings etc. BUT that would mean you cant use the existing top cover, since the motor will "block" the corner. Any thoughts?