VzBoT3D/VzBoT-Vz330 #17: The back plate with no tensioner.stl is not suitable for new EVA 2.4.2 models

arieslight, 2022-01-24:
[image]
[image]

carloshespanhol, 2022-01-24:
I am planing to use this part and have the same problem

simonvez, 2022-01-24:
we will provide one . There is already one if you look in the pull request. A member has made one. We will review it and publish it if it is good

carloshespanhol, 2022-01-25:
In the pull request I found one with tensioner, I' m looking for the back without tensioner.

arieslight, 2022-01-25:
yeah, In the pull request there is no one without tensioner. I modified one without tensioner but it's not perfect

arieslight, 2022-01-25:
[image]
[image]
[image]

arieslight, 2022-01-25:
[plate with no tensioner（beta by arieslight）.zip](https://github.com/VzBoT3D/VzBoT-Vz330/files/7934921/plate.with.no.tensioner.beta.by.arieslight.zip)
it is not perfect,but If you can't wait for the official release，you can try it

closed 2022-01-25 by arieslight