VzBoT3D/VzBoT-Vz330 #37: I dont use social media and I have questions

dkuchay, 2022-07-07:
1.5 years into owning a Voron and I just noticed your project. I dont use social media so the concept of a custom 3d printer is challenging for me due to lack of an ability to ask a question. Can this printer do multiple materials or is the indirect drive going to limit to non TPU based filaments?

chrishenze, 2022-07-09:
Best place to ask question is on discord, there a lot of helpful people. A Vz is intended for very fast prints, but you can modifiy it as you wish. The print head uses direct drive with for example the lgx lite or the Vz-Hextrudort, so you can print most materials except hight temp engineering materials.

dkuchay, 2022-07-09:
Very much appreciate your reply. Thank you. I’ll continue to dive in to VzBot. Thankfully enough resellers carry parts for Voron I can stay abreast of things that users contribute. About the only drawback so far for me. There begins the breakage of the Voron experience though. Stock build sucks and a slight sour taste may result seeing how many mods it takes to get somewhat reliable. I’ve spent 5 years in industrial robotics and attracted to VzBot via gantry and simplicity. Discord more of a won’t for me. Too many bad actors there. Protecting our kids from it but having to use it for other reasons ourselves makes little sense. How cumbersome is the github discussion board? I see it as Vorons mistake for not using. Could have been great repo for mods and support for. Considering their engineers are also authors it again makes me think of a bad taste. IRC chat would even be better than discord for the automation. Lurk, record and filter. Find relevant information quickly. Discord for you is easy. For the user there is a rabbit hole factor where for one, I won’t use public chat for individualized support. Two I just don’t have the time to read each conversation that takes place upon the project in search of my needle within the haystack.  If I were in your shoes it would be an easy choice. I’ve spent 5 years assembling high speed assembly systems aka production lines,  that were sold to pharma. These included robotics, plc’s, vision systems and more. I may have viable ideas to contribute.  Wouldn’t you appreciate a report upon who else wouldn’t even take the time to contact you because they don’t agree with the support venue? Discord may repel those with greater levels of real world experience from going as far as I did as there is no way to ask a question to your project. 

If the discussion board is not to your liking then fire it back at github and feel free to include this message. Microsoft owns them now and they are doing their own dev upon. Demand a better feature. They owe the community for all they did during the 80’s. Watch Netscape docu if you’re too young to remember. Discord is a no go as is FB as is twitter. I know it is for me and i just don’t know how many others I speak for. Do you?

Armoth1, 2022-07-19:
Totally agree with [dkuchay]. I real pity we cant get support on a non-public platform. There is a minority that makes this real unpleasant.

Runningtarrens, 2022-09-03:
The primary community spots are facebook and discord. I use discord. Its a very nice platform, the vzbot server is very nice and full of good people. Over 4k user and fighting at all. I dont understand what you mean, when you talk about non public platform, is this here a non public platform? why are you okay writing here but not on discord? discord is widely used not only for 3d Printer builder but also for vendors, its the best way to get in contact with them. so team would really hamper themselfs without the platform.

anyway, if you do not want to use Facebook or discord is fine, but the difficulty of the build is gona be a bit higher because you might need to figure things out on your own term. But as options there is the git with all neseccary parts uploaded, a complet CAD file. Which you can explore, disassemble and manipulate to your liking, even the usermods that where created and shared on Discord get uploaded into the Usermod repo here on git. There is a manual for the assembly of the printers and the most important settings. Building on top of this there are a lot of videos online on youtube, that shows part of the build, upgrades,configurations, tests and capabilities.

Documentation could alway be better, but please dont forget that its a young project and its only a handful of devs that work on the project. its no where near the size of the voron project/community. So if you still have questions and you feel like you have explored all your options, open an issue with a question here, it might take some time, but people will help.

Armoth1, 2022-09-06:
Hey,

Thank you for your reply. My account got hacked and some bad things
happened from there(FB) so since then I’m very wary. So I stay away and use
email to communicate mostly.

Please note I’m not trying to be difficult or in anyway negative about
VZbot and/or the community, quite the contrary.

Best,

ganzgustav22, 2022-10-28:
This is something I find strange about the VzBot (and other 3dprinter projects). Why is there no forum somewhere? I've had a few hobbies the past years, so far every hobby had one or more forums where people would post their builds, questions, discussions etc. etc.

Now with 3D printers, there are some forums but they seem to be either deserted or populated mostly by peoply who print with 50mm/s and don't want to print faster.

Discord is just a chat, not really suitable for buidlogs and more complex discussions. Where are these things happening?

dkuchay, 2022-10-28:
I posed this question to GitHub wondering the exact same thing and it turns out every GitHub project can elect to activate a discussion board. GitHub suggested that I ask the project lead to activate this or to contact GitHub for help if they didn’t know how to do this. In stead this project along with other 3d printer projects knowingly omit the github in-project offered discussion board. Based on responses this won’t change. Since realizing this I’ve lost interest and have assumed developers to be incapable Sent from my iPhone

simonvez, 2023-01-04:
Discord is way more than just a chat platform. It has FORUM type of posting now and much more. I dont have time and energy to support a Forum too. Discord , FB, Youtube, Instagram is already taking too much of my time :) Discord is probably the best plate for all your need. And like I said, it has evolved a lot with nice addons that you dont need a forum anymore

simonvez, 2023-01-04:
oh.. and we have build logs forum now ;)

closed 2023-01-09 by simonvez