VzBoT3D/VzBoT-Vz330 #39: No motor

Armoth1, 2022-07-21:


Armoth1, 2022-07-21:
Hi guys, just a fyi - re no motor mounts listed on BOM. Would be nice to add and show quantity needed and variants if applicable based on different build options

Runningtarrens, 2022-09-01:
motor mounts are printed. In the BOM is no list for the printed parts, its intended for the parts you need to buy.

Armoth1, 2022-09-03:
Got it thanks. What I'm looking for though is a place where you can see every part of the printer in one clean list like the BOM sheet. So that I cann easily tick off what I have (printed parts and BOM parts) as I'm finding it hard to keep track..  Perhaps not a problem, for most. Just thought I'd share it anyway. Thx for responding its much appreciated.  I'll look up the parts and put together a list for myself by combining BOM list with printed part names, not intending to make more work for you guys. Thx!

Runningtarrens, 2022-09-03:
you could make a copy of the BOM for all the parts that are not printed and then copy the lists of printed parts out of the manual. its not one big list but a handful of smaller, categorized in groups. maybe that would help you getting all the needed parts in one list.