VzBoT3D/VzBoT-Vz330 #49: AWD trouble at slow speed

trackhacs, 2023-03-20:
I have AWD configuration with TMC5160. It works beautiful for normal printing, without any problems.

However, at (very) low speeds  and low acceleration, e.g. during klipper tuning_tower test with suggested acceleration of only 500,  one of the TMC is reporting "no load" and klipper chickens out. 

Guess this is due to both motor pulling in same direction and that might cause one motor to see no load. 

any experience with this?

simonvez, 2023-03-28:
I have personally never heard of that. Did you do the sync procedure ?

trackhacs, 2023-03-31:
yes - but I did only loose one motor per belt. just saw your (?) video about it and will try to loosen all 4 motors.

simonvez, 2023-04-01:
I do only 1 motor per belt too..but doing all 4 is better...just way harder

closed 2023-08-04 by simonvez