VzBoT3D/VzBoT-Vz330 #50: Swapped file names

porkcube, 2023-03-23:
Looks like [exhaust middle cover 10mm Hepa filter.stl](https://github.com/VzBoT3D/VzBoT-Vz330/blob/master/Assemblies%20BOM%20and%20STL/enclosure/Exhaust%20filter%20-%20Fume%20extractor/Exhaust%20filter%20housing/STLs/exhaust%20middle%20cover%2010mm%20Hepa%20filter.stl) and [exhaust middle cover 15mm Hepa filter.stl](https://github.com/VzBoT3D/VzBoT-Vz330/blob/master/Assemblies%20BOM%20and%20STL/enclosure/Exhaust%20filter%20-%20Fume%20extractor/Exhaust%20filter%20housing/STLs/exhaust%20middle%20cover%2015mm%20Hepa%20filter.stl) files are swapped (e.g. 10mm is actually the 15mm / 15mm is actually the 10mm).

simonvez, 2023-03-24:
I think they are ok. the thicker 15mm filter will require a thinner stl part.

closed 2023-03-28 by simonvez

porkcube, 2023-03-31:
OH I see now, I misunderstood the `10mm` / `15mm` are for the filter's size and I was just comparing the models themselves - they're correct as is.