VzBoT3D/VzBoT-Vz330 #54: Printed Y gantry bottom has thin layer in nut hole

utubosk8er, 2023-05-12:
Haven't printed this part yet (Y_gantry_bottom_left.stl), but I do believe the center of a hole that nut sits should be a thru hole, otherwise it would block the tip of M5x30mm bolt from going through.
Is this something that needs to be drilled out or is this how its supposed to be?

[image]
[image]

jaarenhoevel, 2023-05-21:
I suppose that is on purpose to make it easier to print. That way the following round perimeters are not floating in the air.
You may want to print it with the flat side touching the build plate tho.

simonvez, 2023-08-04:
Yes its to avoid printing support. Need to punch through or drill through

closed 2023-08-04 by simonvez