VzBoT3D/VzBoT-Vz330 #55: step opening problem

Moreno-sky, 2023-05-18:
i have a problem during the opening of the step file, my CAd crash near the end of the transformation

magnificu, 2023-07-03:
I have the same problem.

simonvez, 2023-08-04:
What software do you guys use?

simonvez, 2023-09-17:
You can try the web version instead. The full cad is huge and you need quite a good pc to load it

closed 2023-09-17 by simonvez

HumbleDeer, 2023-09-22:
One solution we can issue, is a CAD model with all step files placed in there separately.

Even better would be SLDPRT exports, as STEP files need to be "converted" from solid body definitions to solid bodies and given surfaces that are calculated at opening, before they're displayed.

EduardoMDSousa, 2023-09-22:
Wy? My old laptop with intel grafics 4gb ran Open The full CAD on Fusion360.