VzBoT3D/VzBoT-Vz330 #58: Boot Video Update for Raspbian

recrudesce, 2023-07-28:
Latest Raspbian version doesn't include a VLC player with mmal support, therefore your boot video details don't work.

However, you can easily make it work as follows

`sudo apt install mplayer`

Then rather than using cvlc in the `splash.service`, you instead use this

`ExecStart=/bin/bash -c "mplayer -nosound -nolirc -vf scale=800:480 -vo fbdev2:/dev/fb0 /home/pi//intro1.mp4" &`

Everything else stays the same.

18Markus1984, 2023-08-29:
I now tried omxplayer, cvlc and mplayer nothing worked :(  Why are there the two // instead of  /home/pi/bootvideo?

recrudesce, 2023-08-29:
Typo, but it's irrelevant.

Check your device (fbdev2 etc) and check that the mp4 is in the path you're referencing.

18Markus1984, 2023-08-29:
I tryed to run it, but KlipperScreen doesn't allow it I think.

pi@peter:~ $ mplayer -vo fbdev2 -fs /home/pi/bootvideo/peter.mp4
MPlayer 1.4 (Debian), built with gcc-10 (C) 2000-2019 MPlayer Team
do_connect: could not connect to socket
connect: No such file or directory
Failed to open LIRC support. You will not be able to use your remote control.

Playing /home/pi/bootvideo/peter.mp4.
libavformat version 58.45.100 (external)
libavformat file format detected.
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x766cfc90]Protocol name not provided, cannot determine if input is local or a network protocol, buffers and access patterns cannot be configured optimally without knowing the protocol
[lavf] stream 0: video (h264), -vid 0
VIDEO:  [H264]  800x480  24bpp  30.000 fps   82.8 kbps (10.1 kbyte/s)
==========================================================================
Opening video decoder: [ffmpeg] FFmpeg's libavcodec codec family
libavcodec version 58.91.100 (external)
Selected video codec: [ffh264] vfm: ffmpeg (FFmpeg H.264)
==========================================================================
Clip info:
 major_brand: isom
 minor_version: 512
 compatible_brands: isomiso2avc1mp41
 encoder: Lavf58.76.100
Load subtitles in /home/pi/bootvideo/
Audio: no sound
Starting playback...
Could not find matching colorspace - retrying with -vf scale...
Opening video filter: [scale]
[fbdev2] Can't put VSCREENINFO: Invalid argument
[fbdev2] Can't put VSCREENINFO: Invalid argument
Movie-Aspect is undefined - no prescaling applied.
[fbdev2] Can't put VSCREENINFO: Invalid argument
[fbdev2] Can't put VSCREENINFO: Invalid argument
[swscaler @ 0x767823e8]bicubic scaler, from yuv420p to rgb565le using C
[swscaler @ 0x767823e8]No accelerated colorspace conversion found from yuv420p to rgb565le.
[swscaler @ 0x767823e8]using unscaled yuv420p -> rgb565le special converter
VO: [fbdev2] 800x480 => 800x480 BGR 16-bit  [fs]
V:  16.5   0/  0 16% 32%  0.0% 0 0

Exiting... (End of file)

18Markus1984, 2023-08-29:
I deleted KlipperScreen because I didn't find an easy way to close it, and Kiauh makes it easy to reinstall. I got the video playing but only when KlipperScreen is deleted, and I have some weird artifacts.

```
pi@peter:~ $ mplayer -vo fbdev2 -fs /home/pi/bootvideo/peter.mp4
MPlayer 1.4 (Debian), built with gcc-10 (C) 2000-2019 MPlayer Team
do_connect: could not connect to socket
connect: No such file or directory
Failed to open LIRC support. You will not be able to use your remote control.

Playing /home/pi/bootvideo/peter.mp4.
libavformat version 58.45.100 (external)
libavformat file format detected.
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x76724c90]Protocol name not provided, cannot determi                                                                                                                                                             ne if input is local or a network protocol, buffers and access patterns cannot b                                                                                                                                                             e configured optimally without knowing the protocol
[lavf] stream 0: video (h264), -vid 0
VIDEO:  [H264]  800x480  24bpp  30.000 fps   82.8 kbps (10.1 kbyte/s)
==========================================================================
Opening video decoder: [ffmpeg] FFmpeg's libavcodec codec family
libavcodec version 58.91.100 (external)
Selected video codec: [ffh264] vfm: ffmpeg (FFmpeg H.264)
==========================================================================
Clip info:
 major_brand: isom
 minor_version: 512
 compatible_brands: isomiso2avc1mp41
 encoder: Lavf58.76.100
Load subtitles in /home/pi/bootvideo/
Audio: no sound
Starting playback...
Could not find matching colorspace - retrying with -vf scale...
Opening video filter: [scale]
[fbdev2] Can't put VSCREENINFO: Invalid argument
[fbdev2] Can't put VSCREENINFO: Invalid argument
Movie-Aspect is undefined - no prescaling applied.
[fbdev2] Can't put VSCREENINFO: Invalid argument
[fbdev2] Can't put VSCREENINFO: Invalid argument
[swscaler @ 0x767d73e8]bicubic scaler, from yuv420p to rgb565le using C
[swscaler @ 0x767d73e8]No accelerated colorspace conversion found from yuv420p t                                                                                                                                                             o rgb565le.
[swscaler @ 0x767d73e8]using unscaled yuv420p -> rgb565le special converter
VO: [fbdev2] 800x480 => 800x480 BGR 16-bit  [fs]
V:  16.5   0/  0 20% 32%  0.0% 0 0

Exiting... (End of file)
``` 

https://github.com/VzBoT3D/VzBoT-Vz330/assets/52409213/cb10a43f-9033-43ff-a7d2-dec8673c063d

stonemandave, 2023-10-01:
I got the first intro video to work but the second doesn't. Do you know how i can convert to mplayer.

su - pi -c "sleep 2 | DISPLAY=:0 /usr/bin/cvlc -f --quiet --no-video-title --play-and-exit --mmal-display DSI-1 /home/pi/bootvideo/intro2.mp4"

HumbleDeer, 2023-10-02:
mplayer is software, not a format or codec.

o1hitman1o, 2024-01-15:
VLC works for me. Just need to apt install vlc

/etc/systemd/system/splash.service
*********
[Unit]
Description=Splash screen
DefaultDependencies=no
After=local-fs.target

[Service]
User=pi
ExecStart=nohup /bin/bash -c "DISPLAY=:0 /usr/bin/cvlc -f --quiet --no-video-title --play-and-exit --mmal-display DSI-1 /home/pi/boot_video/boot.mp4" &
StandardInput=tty
StandardOutput=tty

[Install]
WantedBy=sysinit.target

*********
I then add the second video to cover more of the booting time.
/etc/rc.local
Add this 
su - pi -c "sleep 2 | DISPLAY=:0 /usr/bin/cvlc -f --quiet --no-video-title --play-and-exit --mmal-display DSI-1 /home/pi/boot_video/boot2.mp4" &

Before the exit 0

Save and reboot

HumbleDeer, 2024-01-17:
It's come to my attention a while back that there's a good chunk of people still running on Debian Buster. Those will have wildly different issues from subsequent versions, especially against the latest. 

An example: A compatibility package present in Debian 11 was removed in Debian 12. This is causing issues with python library access and what libraries to use in regards to resonance measurements.

The issues coming forth from this are still undocumented as of now. 

Could you all please list your specific distribution, version, and version numbers? Use `uname -a` to get all the identifying information.

o1hitman1o, 2024-01-17:
Linux mainsailos 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

Treintjes, 2024-01-25:
thank you! this was the only way to git working on 64 bit mainsail installation! worked perfectly

closed 2025-01-02 by pbsuper