VzBoT3D/VzBoT-Vz330 #62: Mellow 330 STL list suggested changes

1980alexGS, 2023-09-15:
I think the items in https://github.com/VzBoT3D/VzBoT-Vz330/tree/master/Assemblies%20BOM%20and%20STL/enclosure/Front%20door%20latches/Magnetic%20latch
should replace the different door latch items in 'Mellow kit - all STLs' - Enclosure/Locking latch, since the kit now includes magnets.

Secondly, in the 'Mellow kit - all STLs', the Back Electronic Bay section contains 'Front cover', 'Back cover', and 'screen mount frame' which I don't think belong there, because they appear in the Screen Mount section - please consider removing them.

Also - optionally - consider removing the 2020 Endcaps from the Frame section (they didn't fit for me and they seem unnecessary because the extrusion ends are coloured black), plus the Foot Spacer needs to be about 3.6mm thick to suit the rubber feet supplied in the kit (perhaps indicate this in the name of the file if it is difficult to change the size of the STL)

Thanks!

EduardoMDSousa, 2023-09-15:
Hi the stl are genéric, for all options , kit, and self sorced frame etc , still the best , is download the full printer CAD, and export , since vez keep all parts updated 1st on cad .

1980alexGS, 2023-09-15:
Ok, but I believe the Mellow 330 STL list.zip file should contain the STLs relevant to the Mellow 330 kit - and that specifically is what I’m trying to help with here.

I understand that STLs in other places can remain generic, i.e. both types of latches, different hole sizes, different display housings etc.

Thanks

simonvez, 2023-09-15:
all done

closed 2023-09-15 by simonvez