VzBoT3D/VzBoT-Vz330 #64: HDMI cutouts for mellow fly LCD would've been nice

yuriy-yarosh, 2023-10-02:
There's a HDMI + jack cutout missing in the mellow LCD [back  cover](https://github.com/VzBoT3D/VzBoT-Vz330/blob/master/Assemblies%20BOM%20and%20STL/electronics/Screen%20Mount/Mellow-Fly%20Version/Back%20cover.stl) mount STL.
Front panel needs a bit of a wider cutout as well.
Maybe worth making a separate HDMI version ?

[image]

HumbleDeer, 2023-10-02:
Is this a different display board from the one the case was made for?

yuriy-yarosh, 2023-10-02:
@HumbleDeer No, it's just the display board has multiple adapters for various interfaces.
Mellow vzbot kits are shipped with a DSI adapter for rPI, but DSI is somewhat rPI exclusive, so it make sense to replace it with a HDMI+USB to FRC adapter (flex ribbon) instead.

closed 2025-01-02 by pbsuper