VzBoT3D/VzBoT-Vz330 #65: Corrupted DXF files of panels

montyx99, 2023-11-05:
I've tried to open in multiple viewers the Back electronic panel.dxf and the Bottom panel.dxf files and I get the following error from all of the viewers:
Error: Unexpected end of input: EOF group not read before end of file. Ended on code.

Please review! Thanks in advance

HumbleDeer, 2023-11-19:
I can now confirm this is indeed the case.

Some files are returning errors in Inkscape and LibreOffice Draw, but not in Fusion360. This is probably due to some file export incompatibility or error. 

I will be investigating the cause.

HumbleDeer, 2023-11-19:
FreeCAD has no issues importing the lines in the drawing, but when it does import, it throws all line sections into a base object. The one "Sketch" F360 sees on the original is now a couple dozen different bits and pieces. I wonder if this has any relation to this issue. 

I can't re-export as DXF due to not having the license for it

HumbleDeer, 2023-11-19:
The Fusion360 loaded file shows the following lump (img1) on the right side holes, which Solid Edge does too (img2). Solid Edge opens the file just fine though.
[image]
[image]

In addition to this, it looks like all lines in the drawings are double-exported due to automatic projection of the parts into the sketch plane.

closed 2025-01-02 by pbsuper